bigjosh/TSL-calibre-MSP #4: Maybe move trigger pin up a little?

bigjosh, 2022-10-28:
Current location just seems wrong and prone to having the pin break off. Maybe a little higher would be better?

[image]

cheewee2000, 2024-07-08:
[image]
i think the offcenter ness is OK. the part is positioned to make the lever centered to the pin.

bigjosh, 2024-07-09:
Hmmm. Ideas:

1. Extend the taper of the pin so that it goes up to the middle of the switch arm bearing surface. Seems like this would reduce the sliding up/down force on the switch arm so it would not break? Nice because no PCB changes. 
[image]

3. Make the pin (or at least a section of the pin near the PCB) out of high permeability metal. Put a magnet on one side of the pin and a reed switch on the other. When pin is in place, the field mostly is diverted though the pin. When the pin is removed, the field goes across the hole and closes the reed switch. Nice because no physical contact between pin and PCB. Maybe slight chance that unit could be triggered by big external magnetic field, but I think we can make that very unlikely if we lay everything out well. 

The switch-failure-on-pull is so demoralizing, we have to fix it!

bigjosh, 2024-08-27:
Considering new defects with the trigger pin switch, best to get rid of it altogether.

How about replacing it with a piece of spring that goes across the PCB gap like this...

[image]

To load the TSL, you have to insert the pin unto the end cap and then insert the PCB into the end cap slot. The pin pushes the spring off of the contact point as you slide it in. When the user pulls the pin, the spring returns back to the contact, closing the circuit and starting the count. Nice side effect is that after the pin is pulled once, the spring is completely out of the way and never touches the pin again when the pin is taken out and reinserted later. The user never even sees the spring unless they disassemble the unit and unscrew the PCB from the endcap. 

The end of the spring and the contact could be gold plated where they touch for extra protection. But remember that this switch only has to close once!

cheewee2000, 2024-08-27:
i love this solution!  So if we go with this trigger, we would load the firmware with the pin already loaded? or do we assemble the cap with the pin inserted after loading the firmware?

bigjosh, 2024-08-27:
We would want to test that both opened and closed states work, and probably the easiest way to do this is from the firmware. 

So maybe...

1. program the firmware into the PCB (tests switch closed)
2. load batteries & mount PCB into endcap with pin inserted in the endcap (tests switch open)
3. assemble pcb+endcap into tube

Note that these steps can have long delays between them.

See any problems with this workflow?

cheewee2000, 2024-08-28:
this should work!