bigjosh/TSL-calibre-MSP #46: Replace trigger switch with something more robust

bigjosh, 2024-12-28:
The trigger switch is not well suited for this application. It is pretty flimsy and the arm moves perpendicular to the pin, so when the pin moves it can bend or break the arm. We also had a bad batch of switches that did not close reliably even when the arm moved correctly.

bigjosh, 2024-12-28:
On solution would be to use a magnetically activated switch and have the pin break a magnetic field. Here is a simulation....

[image]

The switch is at the top, a magnet is at the bottom. You can see that when pin is inserted into the hole on (left side) that it redirects the field away from the switch on top. When the pin is pulled (right side), the fields makes it up to the switch which would close in the presence of the field. 

The armatures on the sides could be made from stamped steel. The pin would need to be made from something with high magnetic perativity, so maybe ferritic stainless steel.

The switch could be a reed switch or something solid state like this...

https://www.digikey.com/en/products/detail/coto-technology/RR132-2E33-551/13907128

The solid state version would use an extra ~100nA while in Ready To Launch mode, but maybe more robust to shock than a reed switch would be. Software could mitigate the risk of a false fire due to shock on the reed switch by requiring the circuit to stay solidly closed for some number of milliseconds. 

A downside is that it might be possible to trigger the TSL using a big magnet next to it. This whole assembly is buried inside the end cap so that would make it hard to get a magnet close enough, but this would need to be tested to see if it is possible to trigger with a practically sized magnet.

bigjosh, 2024-12-28:
A simpler solution would be to make a physical arm and would get held by the pin. When the pins gets pulled, the arm is released and is pushed by spring action into the contact. 

[image]

This setup has an added benefit that you can visually see if the pin was ever pulled. Although a user could potentially reset it, that would require knowing that this is what is inside, the right hex wrench, and some effort. 

The arm could be made from wire or stamped. It should probably be gold plated - or at least the part that makes contact. If it is wire, you'd probably want a couple of bends at the joint so that it could be mounted into holes in the PCB? It probably makes sense for the far end to be constrained, maybe with something like this...

[image]

https://www.digikey.com/en/products/detail/harwin-inc/S1621-46R/2264455
 
Not sure what would go on the other end for the arm to make contact with, but could be almost any SMD part with an exposed conductive surface. Maybe a simple existing gold plated part like this?...

[image]

https://www.digikey.com/en/products/detail/iriso-usa-inc/IPS-4103T-01Y900/14291321

cheewee2000, 2025-01-22:
how's something like this...
(shown in its fabricated state)
So there's no load on the spring while it's getting soldered.
When the cap+pin assembly is joined with the PCB, the pin pushes the spring past the post.
When the pin is pulled, the spring moves back to hit the post.

[image]

bigjosh, 2025-01-23:
Yea, i hadn't considered the difficulty of soldering while under tension!

This looks like a very cool solution, and super simple to make. We should test some!

I really like the latchy-ness of it, but I *really* want to to latch when the trigger is pulled rather than (or in addition to) when it is loaded so that the pull is *physically* one-time only.  Can you imagine any ways that could work? If not, then this is great!

cheewee2000, 2025-01-24:
updated the post side so the bend is perpendicular to the spring. The spring is longer and less deep so the spring can be more springy. 

The way I think of it, the latch gets loaded by the pin when it doesn't have any power, so technically the only time it latches is when the pin is pulled.

[image]

bigjosh, 2025-01-26:
❤️

1. Can you check to make sure the center of gravity of the bigger piece falls inside the solder pad?
2. Maybe can we add an extra inward bend at the top of the little hook? Just in case the long arm binds against the pin, this will keep the long arm piece from deforming upwards and ensure that- no matter what- the two arms will contact and trigger the TSL.  

[image]

cheewee2000, 2025-01-26:
you read my mind! I was thinking about this while going to sleep last night. I added a bunch of bends to help with pin binding and adjusted for the center of gravity.

[image]

bigjosh, 2025-01-29:
Looks amazing! Do you think that arm deflection is not too much? Hopefully can be manufactured cheaply even with all those bends! :)

bigjosh, 2025-01-29:
Maybe also try to ask a PCB DFM person if that is going to be easy to place and reflow? Two more bends and there could be alignment tabs on the bottoms.